BCDA-APS/bluesky_training #5: start_adsim.sh and remove_container.sh cannot be accessed in

l123173, 2021-05-31:
Could not get the files in bluesky_train install.md: the start_adsim.sh and  remove_container.sh

Here is the content:
cd ~/bin
wget https://raw.githubusercontent.com/prjemian/epics-docker/master/n4_areaDetector/start_adsim.sh
wget https://raw.githubusercontent.com/prjemian/epics-docker/master/n3_synApps/remove_container.sh

prjemian, 2021-06-01:
Thanks for the report.  Yes, they moved recently.

prjemian, 2021-06-01:
context (in *main* branch):

```sh
(bluesky_2021_2) prjemian@zap:~/.../BCDA-APS/bluesky_instrument_training$ git grep n4_area
describe_instrument.ipynb:    "`ad:` | area detector IOC | [`prjemian/synapps-6.1-ad-3.7`](https://hub.docker.com/r/prjemian/synapps-6.1-ad-3.7) | https://github.com/prjemian/epics-docker/tree/main/n4_areaDetector\n",
install.md:    wget https://raw.githubusercontent.com/prjemian/epics-docker/master/n4_areaDetector/start_adsim.sh
(bluesky_2021_2) prjemian@zap:~/.../BCDA-APS/bluesky_instrument_training$ git grep n3_syn
describe_instrument.ipynb:    "`gp:` | general purpose IOC | [`prjemian/synApps-6.1`](https://hub.docker.com/r/prjemian/synapps-6.1) | https://github.com/prjemian/epics-docker/tree/main/n3_synApps\n",
install.md:    wget https://raw.githubusercontent.com/prjemian/epics-docker/master/n3_synApps/remove_container.sh
install.md:    wget https://raw.githubusercontent.com/prjemian/epics-docker/master/n3_synApps/start_xxx.sh
install.md:    wget https://raw.githubusercontent.com/prjemian/epics-docker/master/n3_synApps/remove_container.sh
```

prjemian, 2021-06-01:
Both branches ([*main*](https://github.com/BCDA-APS/bluesky_instrument_training) and [*4-refactor-one-folder*](https://github.com/BCDA-APS/bluesky_instrument_training/tree/4-refactor-one-folder)) updated.

closed 2021-06-01 by prjemian