BCDA-APS/bluesky_training #86: Bluesky_RunEngine_md default path

rodolakis, 2023-03-28:
Bluesky creates a Bluesky_RunEngine_md directory in Home directory at startup. Let user choose location of directory.

rodolakis, 2023-03-28:
In `iconfig.yml` add:

```yml
RUNENGINE_MD_PATH: /home/beams/RODOLAKIS/data/RE_md
```

In `instrument/framework/initialize.py` replace 

https://github.com/BCDA-APS/bluesky_training/blob/c52bd6b175e34e9c8ad938598e3c4613e80485de/bluesky/instrument/framework/initialize.py#L44-L66

with:

```python
def get_md_path():
    return str(pathlib.Path(iconfig["RUNENGINE_MD_PATH"]))

# Set up a RunEngine and use metadata backed PersistentDict
RE = RunEngine({})
RE.md = PersistentDict(get_md_path())
```

prjemian, 2023-03-28:
We should also add handling in case the user does not provide this key (it is commented out, by default).

closed 2023-03-28 by rodolakis