BCDA-APS/bluesky_training #100: Simpler Hello, World notebook

prjemian, 2023-04-19:
The [Hello, World](https://bcda-aps.github.io/bluesky_training/instrument/_hello_world.html) example installs the full `instrument` package.  It could be simpler, installing just enough (RunEngine and a databroker temporary catalog) to make the notebook work.

prjemian, 2023-04-19:
Note that `quick_hello` is used in the _Hello, World_ notebook.

prjemian, 2023-04-19:
Cell `In [1]` might be replaced with:

```py
import pathlib, sys

path = str(pathlib.Path("/tmp") / "bluesky")
pathlib.os.chdir(path)
sys.path.append(path)

import bluesky
import databroker
from bluesky.callbacks.best_effort import BestEffortCallback

cat = databroker.temp()
RE = bluesky.RunEngine()
RE.subscribe(cat.v1.insert)
RE.subscribe(BestEffortCallback())

```

and then eliminate the discussion of logging and its details.

prjemian, 2023-04-19:
Also, there is typo on this line:

> The data from this run is available until we quite the IPython session

Should use `quit` instead of `quite`

rodolakis, 2023-04-20:
I copied the HelloWorld from /instrument into /turor and added some more detailed explanations (some hidden in pull down menus). The original HelloWorld can maybe be combined to the "Test the New Bluesky Instrument"?
When I tested the notebook to refresh the output, I had to add os.chdir(my_path) since I was in a random directory and %run -i user/quick_hello.py failed. There maybe a more elegant way to do this with pathlib but I could not find it easily.

rodolakis, 2023-04-21:
I just realized that the previous version did include a:
`$ cd /tmp` 
(should have been home instead, as discussed earlier") but that was in the "Installation step" pull down menu, which is why I missed it! I am thinking I want to make 
`$ cd ~/bluesky`
part of the "start bluesky session" page I mentioned in: 
- #104 
then link that page at the start of HelloWorld.

rodolakis, 2023-04-21:
@prjemian which makes we wonder, can I "retract" a pull request? (if I want to make more changes in my #100-HelloWorld branch?)

rodolakis, 2023-04-21:
Updated comments from last night.

rodolakis, 2023-04-21:
Figured out how to cancel my pull request and resubmit, I replaced ```os.chdir(my_path)
%run -i user/quick_hello.py ```

by 

`%run -i ~/bluesky/user/quick_hello.py`

prjemian, 2023-04-21:
> can I "retract" a pull request

Yes, in a couple ways.  One way is to mark it as a _draft_, which means do not review or merge it yet.  (Implies there is more work that will happen.)

Another way is to close the pull request without merging.  This does not delete the branch, just cancels the intent to review and merge the branch _at this time_.  It's the most common way to indicate no more work is being done on this PR.  Leave a comment in the PR to describe why.  Can always create a new PR for a branch.

Another way is to delete the branch, at which time the PR probably self-destructs since there is, then, no branch to be merged.  GitHub might not let you delete a branch (I don't know) which is still part of a PR.  The usual process is close the PR(s) first, then delete the branch.

prjemian, 2023-04-21:
> `%run -i user/quick_hello.py` failed

You are right.  This fails since it tries to run using a file reference.  Because the reference is a _relative_ path (which assumes some starting directory), it failed.  If an absolute path is used (such as `/home/somebody/bluesky/user/quick_hello.py`, this will not be general.

I agree the pragmatic correction is to use `os.chdir()` (or `path.os.chdir()` to avoid `import os`)

An alternative is to use `%run -im user.quick_hello` (loads a Python _module_ instead of a Python file)  which assumes:

- the `user` directory is in the `sys.path` (which is true)
- the `user` directory has an [`__init__.py`](https://bcda-aps.github.io/bluesky_training/instrument/_about_init_files.html) file (which could be added)

Might be the better way.  I know that the difference between `%run -i file.py` and `%run -im module` was not so easy to learn.  I like the module way better.  Our SPEC users are used to `qdo macrofile.mac` in SPEC where the macrofile.mac is a file name (with path if needed).

PR #105 (2023-04-27): 100 hello world
- close #100
closed 2023-04-27 by rodolakis