BCDA-APS/bluesky_training #116: try out the haven device registry

prjemian, 2023-04-26:
https://github.com/spc-group/haven/

pip install poetry
use poetry to install haven

prjemian, 2023-05-24:
https://stackoverflow.com/questions/66422871/how-do-i-install-python-packages-with-poetry

prjemian, 2023-05-24:
```bash
(bluesky_2023_2) prjemian@zap:~$ poetry add haven

Poetry could not find a pyproject.toml file in /home/prjemian or its parents
```

prjemian, 2023-05-24:
skip `poetry` and pip install from working directory:

```bash
mkdir -p ~/Documents/projects/spc-group
cd  ~/Documents/projects/spc-group
git clone https://github.com/spc-group/haven
cd haven
pip install -e .
```

prjemian, 2023-05-25:
```py
from haven import registry
registry.register(EpicsSignal)
registry.register(EpicsMotor)
registry.register(scaler)
registry.register(fourc)
registry.register(shutter)
```

prjemian, 2023-05-25:
Can add `haven` to the environment spec with a requirement similar to this one: https://github.com/BCDA-APS/bluesky_training/blob/7620c1d95b075bb74209e26f6008547aa57c384d/bluesky/environments/environment_2023_2.yml#L118

These additions to the yaml file are needed to use the `haven` instrument registry as-is:

```yaml
  - mergedeep
  - xraydb

  - pip:
    - https://github.com/spc-group/haven/archive/main.zip 
```

closed 2023-05-25 by prjemian

canismarko, 2023-07-01:
I published Haven to the PyPI: https://pypi.org/project/haven-spc/

Fair warning, the [gui] extras does not currently resolve, there's a dependency hell situation with PyQT apparently. It should work without that extra though.

prjemian, 2023-07-01:
Great!