BCDA-APS/bluesky_training #129: DOC Hello world notebook has extra cells in documentation

rodolakis, 2023-04-28:


prjemian, 2023-05-10:
Seems to be a common problem for all notebooks.

prjemian, 2023-08-21:
I can replicate this behavior in a local build.

prjemian, 2023-08-21:
In the editor:
[image]

After rebuilding the docs, here's the view in the browser:
[image]

prjemian, 2023-08-21:
Which uses this HTML code:

```html
<p>What about extra cells?</p>
<div class="nbinput docutils container">
<div class="prompt highlight-none notranslate"><div class="highlight"><pre><span></span>[1]:
</pre></div>
</div>
<div class="input_area highlight-ipython3 notranslate"><div class="highlight"><pre><span></span><span class="nb">print</span><span class="p">(</span><span class="s2">&quot;Hello&quot;</span><span class="p">)</span>
</pre></div>
</div>
</div>
<div class="nboutput nblast docutils container">
<div class="prompt empty docutils container">
</div>
<div class="output_area docutils container">
<div class="highlight"><pre>
Hello
</pre></div></div>
</div>
```

prjemian, 2023-08-21:
This `div` block is responsible: `nbinput docutils container`

[nbsphinx](https://nbsphinx.readthedocs.io/en/0.9.2/)

prjemian, 2023-08-21:
Apparently, the order of listing the Sphinx extensions is important.  Problem goes away when this one is moved to the end of the list: https://github.com/BCDA-APS/bluesky_training/blob/9bc5b9e83bddbb46fbc4af02bc82767afaf08c4f/docs/source/conf.py#L27

prjemian, 2023-08-21:
[image]

PR #234 (2023-08-21): add sphinx_tabs.tabs extension last
- close #129
closed 2023-08-21 by prjemian