BCDA-APS/bluesky_training #131: DOC git advice for the reference section

prjemian, 2023-04-28:
With a new account, the default git user and email are not set.  These are the [steps](https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-email-preferences/setting-your-commit-email-address):

```bash
git config --global user.name "Lynn User"
git config --global user.email "YOUR_EMAIL"
```

At the same time, extend the interval when you don't have to retype your github [credentials](https://git-scm.com/docs/git-credential-cache) (a 1 hour period is shown):

```bash
git config credential.helper 'cache --timeout=3600'
```

prjemian, 2023-04-28:
Seems to be a FAQ.

prjemian, 2023-04-29:
Also for the git help document.

rodolakis, 2023-08-02:
> With a new account, the default git user and email are not set. These are the [steps](https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-email-preferences/setting-your-commit-email-address):
> 
> ```shell
> git config --global user.name "Lynn User"
> git config --global user.email "YOUR_EMAIL"
> ```

The error message for this is very straightforward, give you those exact instructions. I don't think we need to explain this. Users should be able to handle it? (it is also easy to find the answer on google)

rodolakis, 2023-08-02:
> At the same time, extend the interval when you don't have to retype your github [credentials](https://git-scm.com/docs/git-credential-cache) (a 1 hour period is shown):
> 
> ```shell
> git config credential.helper 'cache --timeout=3600'
> ```

SSH keys allow you to establish a secure connection between your computer and GitLab/Hub, so you never have to authenticate. I am thinking that authentification preferences might be outside of the scope of our documentation?

PR #218 (2023-08-07): 141 git instructions
- close #141 
- close #131
closed 2023-08-07 by rodolakis