BCDA-APS/bluesky_training #139: DOC _testing.md

rodolakis, 2023-05-02:
Name is not very meaningful. 
Contain both:
- ophyd testing with S:SRcurrentAI => redundant with _test_bluesky_at_APS.md
- APS data management   => could be a separate file (not every beamline are setup for the DM system)
I want to re-purpose _testing.md to only address the APS data management, and merge the ophyd part with _test_bluesky_at_APS.md

prjemian, 2023-05-02:
OK.  These document was intended to test during the installation process, that certain software packages were installed and function as expected.  The entire content of this file could be relocated.

rodolakis, 2023-05-03:
Re-purpose to APS data management system only, and relocated to reference

PR #143 (2023-05-10): Installation guide review
- close #112
- close #139
- close #119
- close #127
closed 2023-05-10 by prjemian