BCDA-APS/bluesky_training #184: add haven to new instrument environments

prjemian, 2023-05-25:
Add the `haven` package to new instrument environments.  See #116 for details.

- #116

prjemian, 2023-08-12:
But haven requires python<3.11:

```py
(bluesky_2023_2) zorinvm@zorin22:~/.../bluesky/environments$ pip install haven-spc
DEPRECATION: Loading egg at /home/zorinvm/Apps/miniconda/envs/bluesky_2023_2/lib/python3.11/site-packages/stdlogpj-1.0.3-py3.7.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
ERROR: Ignored the following versions that require a different python version: 0.1.0 Requires-Python <3.11,>=3.7; 23.7.1 Requires-Python <3.11,>=3.7
ERROR: Could not find a version that satisfies the requirement haven-spc (from versions: none)
ERROR: No matching distribution found for haven-spc
```

prjemian, 2023-08-21:
Needs more thought.

prjemian, 2023-08-21:
An alternative to adding the `haven-spc` package (and its requirement of `py <3.11`) is [vendoring](https://stackoverflow.com/questions/26217488/what-is-vendoring#39643873) its [`InstrumentRegistry`](https://github.com/spc-group/haven/blob/eb5efdcd44554a07cad70acc3147edac7e8fe851/src/haven/instrument/instrument_registry.py#L32) support.

@canismarko -- Thoughts?  Can the requirement be relaxed?

canismarko, 2023-08-21:
Unfortunately, Haven is stuck on python 3.9 until this pydm issue gets resolved: https://github.com/slaclab/pydm/issues/951

I *could* allow python 3.11, but I can't run tests on CI due to the above-mentioned issue, so I'd rather not.

I re-factored the instrument registry so that it has no other Haven dependencies, so vendoring it might be the best choice for now. 

https://github.com/spc-group/haven/commit/4e066d644c447dff5d57b5e3a10495b631139492

prjemian, 2023-08-21:
Understood.  Thanks, @canismarko !

prjemian, 2023-08-30:
How to register any device that uses a specific class?

prjemian, 2023-08-30:
Thanks, you updated the comment above.

PR #245 (2023-09-25): add ophyd-registry to new environment for 2024_1
- close #184
Thanks, @canismarko for this hoist from the haven package!
closed 2023-09-25 by prjemian