BCDA-APS/bluesky_training #197: DOC IPython instructions need more details

prjemian, 2023-06-06:
These [instructions](https://bcda-aps.github.io/bluesky_training/reference/_ipython.html#add-a-single-python-file-to-the-startup-directory) need more detail.  As they appear now, they show the resulting file structure.

rodolakis, 2023-06-06:
There are already instruction to create a python profile in the new instrument installation documentation:

https://bcda-aps.github.io/bluesky_training/instrument/_install_new_instrument.html#create-a-bluesky-ipython-profile

prjemian, 2023-06-28:
The instruction to create a new ipython profile: `ipython profile create bluesky --ipython-dir="~/.ipython"` should use an environment variable to specify which directory.  The [instructions to be removed](https://bcda-aps.github.io/bluesky_training/reference/_ipython.html#create-ipython-profile-for-bluesky) define the _wrong_ environment variable.  The correct version should be:

```bash
export IPYTHON_DIR=~/.ipython
# export IPYTHON_DIR=~/.ipython-bluesky  # an alternative
mkdir -p "${IPYTHON_DIR}"
ipython profile create bluesky --ipython-dir="${IPYTHON_DIR}"
```

The expected definition of `BLUESKY_DIR` is the location of the directory containing the `instrument` package.  Typically:

```bash
export BLUESKY_DIR=~/bluesky
```