BCDA-APS/bluesky_training #242: Plan how to update a beamline instrument package

prjemian, 2023-08-23:
Updating a beamline's `instrument` package is difficult, involving lots of expert decision making.  Probably beyond what we can expect from instrument teams.

Make this process easier.  It will become a routine activity and should not be painful or beamline's will be reluctant to upgrade.

@cpchuang Thanks for the prompting on this!

MDecarabas, 2024-04-22:
[image]

prjemian, 2024-04-22:
With some redesign of this repo, including moving into subrepositories, upgrades could be come simpler.  In the drawing above, the blue boxes are the parts the user is not expected to change and could be upgraded by automation.  The red boxes are created on the initial installation and then not upgraded except by the user.

prjemian, 2024-10-30:
Looking for the copier-template to enable this capability. See this repo: https://github.com/BCDA-APS/bs_model_instrument

prjemian, 2025-01-21:
See this howto guide: https://github.com/BCDA-APS/bs_model_instrument/pull/38

closed 2025-01-21 by prjemian