BCDA-APS/bluesky_training #283: Make installation checklist and installation guide header consistent

rodolakis, 2024-02-28:
Some users were confused about the checklist at the top of the installation guide: should they follow checklist or the installation guide content? The wording of the TOC for the installation guide should be consistent with the working of the checklist, and ideally, the checklist link would link to the section from the installation guide (which can then refer to an external link).
The checklist most likely has the correct order of operation, so we should make the guide match the checklist (not the other way around).

prjemian, 2024-04-10:
It helps if the checklist uses rst anchors to reference the various steps.  Sphinx will pull the exact text from each anchored section and use it in the checklist.

### example

```rst
#. :ref:`instrument.create_bluesky_directory`

.. _instrument.create_bluesky_directory:

Create bluesky directory
------------------------
```

This makes an entry with text  _Create bluesky directory_ (which is linked to the anchor point).

prjemian, 2024-04-10:
We discussed ...

> Could move each section to its own file.

PR #289 (2024-04-29): home page should look friendlier
- close #283
- close #284
- close #285
- close #288
- close #295
- close #298
closed 2024-04-29 by prjemian